raspberry-sharp/raspberry-sharp-io #24: Has anybody managed to make the Dht22 code working (Raspberry.IO.Components.Sensors.Temperature.Dht)?

InTheMerde, 2015-02-02:
Hi

Been trying using different timing, even w/out using the GpioInputOutputBinaryPin class (directly through the driver), still not working (blocking at the time where the host is waiting for the Dht22 to lower the pin after the host issues a low (start signal) and a short high (40us)).
I confirm the sensor is working, as I can read the temp and hum using python code on the same pin.

Thx

ebezine, 2015-03-13:
I pushed a new version that work with DHT11. It should also work with DHT22 without change.

Be carefull, this require to use a MemoryGpioConnectionDriver. Other implémentations are not fast enough.

gamondue, 2015-03-14:
Actually it really needs a small change, because a DHT22 has a different bit encoding. 
I started form your code and I am experimenting. It "almost" works, meaning that I get MANY MANY errors, but when it works the result are right. 
Do you need to do something special to use the MemoryGpioConnectionDriver or simply using your code will do?

ebezine, 2015-03-14:
Hello,

Thank you for your feedback. Feel free to rename the DhtConnection into Dht11Connection and create a Dht22Connection for the DHT22.
In order to use the MemoryGpioConnectionDriver you have to instanciate it explicitely, as it's done in the Test.Gpio.DHT11 test project.

I'm a bit worried about the fact you get many errors : the DhtConnection currently allow 10 retries, do you mean that you may get more than 10 errors ? What kind of Raspberry hardware do you use ?

Also, maybe you're using a pullup resistor : this is something I stopped to do after reading https://www.virtuabotix.com/virtuabotix-dht22-pinout-coding-guide/

Regards,
Eric

gamondue, 2015-03-14:
If I don't wait between retries, I get some "ten retries errors". I had tried with an earlier version of my modified test program and obtained these values (no wait between tries): 
No pullup: 
Measurements 1065, Errors 22, Error rate 2,07%, TotalRetries 1836, Retries rate 17,24%
10 kohm pullup: 
Measurements 1662, Errors 51, Error rate 3,07%, TotalRetries 3464, Retries rate 20,84%
Error is "10 times retry", retries rate is <Total number of retries>/<measurements * 10>. 
Since the producer states that it would be better to wait at least 2 s between samples, I put a wait in my DHT22Connection code that enforces such wait. 
With such wait the error rate has become almost zero but retries still persist (I have no statistics, nor extensive test still, because it takes MORE (and MORE) time).

ebezine, 2015-03-14:
Hello, thank you for your feedback.
I'll do some benchmark also on my side (with DHT11 though)

Which revision of Raspberry Pi do you use ?
Regards,
Eric

gamondue, 2015-03-14:
I am on a plain B version 2. 
The reliability seem to be quite "volatile". If it starts bad it really continues bad. If it starts good it often remains good. I read somewhere that the chip tends to drift and, if I remember right, "lock" when autoheats. Maybe one could switch the chip off when not using it and and feed it just a moment before measurement. I saw a Python program that was doing so. 
I pushed on my fork in GitHub my latest version, with testing provisions.

ebezine, 2015-03-15:
Hello @gamondue : I pushed some updates concerning DHT:
 - Splitted DHT11 / DHT22 using a base class for protocol and virtual members for computation
 - Improved timings in GpioMemoryConnectionDriver, and added ability to load a driver based on capabilities (see DHT11 sample).
 - Added ability to keep the DHT active or not between measures.

I updated the protocol for DHT22 based on code in your branch as well as DHTlib (https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTlib), please test it (I don't have a DHT22).

I made some tests on my DHT11 and there is almost no difference whatever the settings (keeping the DHT active or not, minimum interval between two measures) : on 38 samples, average number of attempts is between 1.8 and 2.2 and standard deviation between 0.9 and 1.5.

It's quite strange you don't have better result as I have, because I still don't have a Raspberry Pi B version 2 and still use a B+.

gamondue, 2015-03-16:
Your (beautiful) code seems to work flawlessy. I get a very low number of attempts and right data. 
Fantastic job! 
P.S. I dont'have a Raspberry Pi 2 B, but a Raspberry Pi "1" version 2, model B, thus a "old" B "without +".

ebezine, 2015-03-16:
Hello @gamondue, nice to see that it's better now!
I just pushed a change enabling to set the minimal sampling interval. Setting it to 100ms (on DHT11) has no visible effect on reliability (but is ways faster !)

closed 2015-03-16 by ebezine

mala-zaba, 2015-06-21:
Hi, I have a Raspberry PI 2 B and DHT22. I try the Test.Gpio.DHT11 code (change Dht11Connection to Dht22Connection) but I unable to read data (Unable to read data). My sensor work with loldht. I did some thing wrong with DHT11 sample code?

BennyM, 2016-07-29:
I'm also seeing strange behaviour on a Pi3. Unable to pinpoint the issue at the moment.

[image]

gamondue, 2016-07-29:
If the error is the checksum, maybe you have something in cabling. I had used the HT22 without a pullup resistor, but maybe your setup needed it. Or maybe the cable is too long, even if the communication should be quite robust. 
The distinction between one and zero is done with the time period of the pulses from the sensor, so it could even be a problem of timings or too many interruptions from the O.S. Try to test with just the I/O program and nothing else, possibly from the shell, with no XWindow (graphical environment).
If you manage to have the sensor working, please report here the solution.

BennyM, 2016-07-29:
I'm using Raspbian Jessie Lite with nothing else running. The cables are only 5 cm, I'm using jumper wires at the moment. It works fine when I use an arduino and the arduino libraries. A plain Pi with mono gives me this results. The pull up resistor gives the same result.

BennyM, 2016-08-10:
I have confirmed that it just doesn't work. I've used a C library and wrapped it so I can use it in mono. There are probably too much timing issues for it to work in a managed environment.