raspberry-sharp/raspberry-sharp-io #27: Can't build with mono/Mac

Savij, 2015-02-22:
Hi,

Just tried compiling on my mac osx 10.10.1 (yosemite). Fails in Raspberry.IO.Interop.GenericIoControl<T>            
 [DllImport("libc", EntryPoint = "ioctl", SetLastError = true)]

error is The DllImportattribute cannot be applied to a method that is generic or contained in a generic type. 

From the microsoft docs:
https://msdn.microsoft.com/en-us/library/system.runtime.interopservices.dllimportattribute(v=vs.110).aspx:

"The DllImportAttribute does not support marshaling of generic types."

I guess the 'T' in public static extern int ioctl(int descriptor, UInt32 request, ref T data); wont work with the DllImport. I have not tried this on a pc yet. I tried it on monodevelop on the pi, and apparently I can't open the solution file there (I think that's because of the older version of monodevelop on the pi). I figured it would be fun to try it on the mac being that it's also a unix distro. 

Jeff

ebezine, 2015-03-13:
Also reported by @gloveboxes in #33

ebezine, 2015-03-13:
Hello @Savij & @gloveboxes. You're absolutely right, generic types shouldn't be used for Marshaling.
This part has been submitted by @danm-de : could you imagine a different way to implement IControlDevice ?

danm-de, 2015-03-13:
Woops. I'll have a look at it.

danm-de, 2015-03-13:
Hello @Savij & @gloveboxes,
Unfortunately that little "cheat" (using a generic inner class for marshalling) does not work/compile anymore. I have just refactored the problematic code:

https://github.com/danm-de/raspberry-sharp-io/commit/965a8a00a699b25e9785e01cbfac3453b4edfd44

NativeSpiConnection uses the SPI interface of your Raspberry PI (19 MOSI, 21 MISO, 23 SCLK, 24 CE0, 26 CE1) and requires the spi-bcm2708 kernel modul loaded. It reads/writes the /dev/spidev* device file using IOCTRL (--> ControlDevice.cs did not compile).

@ebezine: I couldn't test it on my Raspberry PI (don't have any SPI device around at the moment)
Because of this I am not sure whether it is "stable" enough for your master branch. 

 :-(

gloveboxes, 2015-03-13:
Hi there, many thanks for the fix, I couldn't see update on any branches.  I can test, have a MCP3002 ADC with light sensor attached.  Can put code somewhere I can download from?  Cheers Dave

danm-de, 2015-03-13:
@gloveboxes I haven't created a pull request yet. It's updated in my forked repository:
https://github.com/danm-de/raspberry-sharp-io.git

Note: the MCP3002 does not use the native SPI interface.

Savij, 2015-03-14:
Hey guys, I have an mcp3008 with a CDs cell. I'll try to test out the branch this weekend and let you know how it goes!! I actually found a better solution than the ADC for my project, I'm using a comparator instead to detect blinking light, but still interested in helping you test the fork out. I'll get back to you soon.

Jeff

gloveboxes, 2015-03-15:
Hey there, not 100% sure what you mean, are you saying mcp3002 no longer supported? Cheers Dave

ebezine, 2015-03-15:
Hi @gloveboxes, MCP3002 is atoll supported but do not use the native SPI interface.
Hence it's not affected by code rework made on native SPI interface.

Regards,
Eric

danm-de, 2015-03-15:
@ebezine: https://github.com/raspberry-sharp/raspberry-sharp-io/pull/35
Tested an working.

Savij, 2015-03-15:
I can confirm the fork builds, but I have not moved the dll's over to the pi to see if everything still works as expected. May take me an extra day to get this done, but it definitely builds which it didn't before.
-Jeff

gloveboxes, 2015-03-15:
Will have mcp3008 in next few days, will also test...

ebezine, 2015-03-15:
Thank you @danm-de, I just merge the pull request.

@gloveboxes : currently only Tlc59711 uses the NativeSpiConnection. MCP3008 also use the SpiConnection instead, which is not impacted by these changes.

closed 2015-03-15 by ebezine

gloveboxes, 2015-03-15:
Ah cool, thought that might be the case. Thanks for the fix, even no direct impact at least I can compile again :)