raspberry-sharp/raspberry-sharp-io #32: i2c & sudo

neutmute, 2015-03-03:
I have some code using i2c which works fine in the console when elevated.

If I use it with winforms, then winforms doesn't work under sudo (get "check your display environment variable" error)

The winforms code not executed under sudo works fine but then I can't get to the i2c.
.
Any thoughts on how to work around this so that I can use i2c under winforms?

neutmute, 2015-03-03:
I just realised one factor might be i am using TightVNC to serve XWindows. Haven't tried this via an HDMI connection. Maybe that will work....hmm.

WildGenie, 2015-03-05:
Are u try gksudo?

neutmute, 2015-03-06:
@WildGenie You sir, should change your name to WildGenius! It worked - thanks!

closed 2015-03-06 by neutmute