raspberry-sharp/raspberry-sharp-io #38: I2C: Repeated start conditions are not supported

chkr1011, 2015-04-20:
These conditions are used to read registers from port expanders like PCA9555. The library does not allow to write and read within the same transaction. This issue also prevents reading from EEPROMs etc.

ebezine, 2015-06-15:
It seem you have solved this issue through Pull request #39.

closed 2015-06-15 by ebezine

JTrotta, 2017-09-25:
It seems this enhancement not working with LTC2943, that needs reapeted Start conditions. Is this the right way to use it?
`            I2cWriteAction writeAction = new I2cWriteAction(register);
            I2cReadAction readAction = new I2cReadAction(new byte[byteCount]);
            I2cTransaction transaction = new I2cTransaction(writeAction, readAction);
            Execute(transaction);

            return readAction.Buffer;`

chkr1011, 2017-09-26:
Hi,
yes this should work. I fixed acessing port expanders like PCA9555 or MAX7311 with this code. But I don't know what heppened in the meantime or of your device requires some other things.

I remember that there were some issues with the clock stretching feature which is not properly implemented in hardware. This results in several devices which are not working. But I don't remember if this was RaspberryPi 1 or RaspberryPi 2.

Best regards
Christian

JTrotta, 2017-09-26:
@chkr1011 Thank you for reply. I'm testing library on RB3, and LTC2943 (fuel gauge sensor), but I can only read all register at time. So you suggest to check the clock?  The only way to do this is using an oscilloscope.
I'll let you know. Thank you very much

P.S. (you MQTT library is very cool!)

chkr1011, 2017-09-26:
No there is nothing you can do and I assume this is fixed on Pi3 (it was a bug in hardware and not fixable in code). 

Thank you :-)

JTrotta, 2017-09-26:
But I'm using Pi3/CM3 (industrial Compute Module 3), and cannot get it working. I suppose there must be something to let it work. This is the sequence:
[image]

JTrotta, 2017-10-03:
I'm sorry, but this PR does not work at all. It does not sends a repeated started condition (oscilloscope checked).
I engineered a new working driver using c++ code from here : [https://github.com/NextThingCo/otto-runner/blob/master/lib/bcm2835.c](url)

Now it works. I'm publishing a PR,
Thank you.