raspberry-sharp/raspberry-sharp-io #53: Button not working as expected on Raspberry Pi B V2

wcom939, 2015-12-26:
Hello,

I wired a simple push button as specified here http://razzpisampler.oreilly.com/ch07.html Whereas the ground is on Pin 39, and the GPIO being used is at Pin 37 (GPIO 26).

The code is as follows:

```C#
var switchButton = ConnectorPin.P1Pin27.Input().Revert().Switch().Enable().OnStatusChanged(x => {
Console.WriteLine("Button Switched {0}", b ? "on" : "off");
});

using(var connection = new GpioConnection(switchButton)){
Console.WriteLine("Init Completed");
Console.ReadKey();
}
```
When pressing the button the text is not getting stated, when manually shorting the wires the text is not getting stated, when using my finger and touching one of the wires the text gets randomly fired.

Please advise on how to fix this problem.

Thanks,
Dan

attilabalazsy, 2016-04-12:
Hi Dan,

I've a similar issue. Some of the pins are working as expected and some are returning random values. I think, that there is a problem in the memory mapping or somewhere else ... maybe there were some changes in PI 2 addressing? Did you find out your issue?

Thanks,
Attila

fahadash, 2016-07-21:
I am having the same issue. My push button even is randomly firing and ignoring my pushes. It fires every so often sometimes with true sometimes with false values. This is really killing us.

attilabalazsy, 2016-07-25:
Do you or anybody is having any clue what is causing the issue? Is it a driver problem? I'm experiencing the problem on some ports only, and some are working correctly.

fahadash, 2016-07-27:
@attilabalazsy I found the solution. It looks like the way button is hooked up to the Pi in most of the examples is incorrect . It may work for older models, but not with my Pi 2 B+.

Connecting GPIO pin to the Ground through the switch doesn't work, I tested using the multimeter and found that GPIO pin was hardly giving out 1 volt, and you need good 3 volt signal. So that connection may have worked on older ones I assume because it is everywhere on internet.

Here is the correct way to connect the switch
http://imgur.com/a/hz8km

fahadash, 2016-08-02:
@attilabalazsy @wcom939  @keneo  @logicethos @shibayan 
I found out that this method is called "pull up resistor" and that if you don't want to wire up that way, you can invoke internal pull-up resistor that RPI provides through code, I don't know how Raspberry-Sharp-Io allows us to enable that resistor.