raspberry-sharp/raspberry-sharp-io #56: I2C not working on Pi 3

StarlightNL, 2016-03-05:
This code works on my Pi 2 but not the new Pi 3
The code and the libs for the project is on my github account
```
pi@Natsukipi:~/monodevelop/BattleBotServer/BattleBotServer/bin/Debug $ sudo mono BattleBotServer.exe

Unhandled Exception:
System.TypeInitializationException: An exception was thrown by the type initializer for BattleBotServer.Helpers ---> System.NotSupportedException: Connector pintout Unknown is not supported
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase (ProcessorPin sdaPin, ProcessorPin sclPin) [0x00000] in <filename unknown>:0
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver..ctor (ProcessorPin sdaPin, ProcessorPin sclPin) [0x00000] in <filename unknown>:0
  at BattleBotServer.Helpers..cctor () [0x00000] in <filename unknown>:0
  --- End of inner exception stack trace ---
  at BattleBotServer.LogWriter.get_Instance () [0x00000] in <filename unknown>:0
  at BattleBotServer.MainClass.Main (System.String[] args) [0x00000] in <filename unknown>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: An exception was thrown by the type initializer for BattleBotServer.Helpers ---> System.NotSupportedException: Connector pintout Unknown is not supported
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase (ProcessorPin sdaPin, ProcessorPin sclPin) [0x00000] in <filename unknown>:0
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver..ctor (ProcessorPin sdaPin, ProcessorPin sclPin) [0x00000] in <filename unknown>:0
  at BattleBotServer.Helpers..cctor () [0x00000] in <filename unknown>:0
  --- End of inner exception stack trace ---
  at BattleBotServer.LogWriter.get_Instance () [0x00000] in <filename unknown>:0
  at BattleBotServer.MainClass.Main (System.String[] args) [0x00000] in <filename unknown>:0
```

OddmarDam, 2016-05-14:
Any news on this ? I also have a RPI 3 that i would like to use instead of my RPI 2.

msanchez71, 2016-05-20:
Is this project still active?  I would like to know when this issue with Pi 3 might be looked at.

Thank you in advance

jyksnw, 2016-05-20:
Looks like a new PinMapping needs to be defined. It is my understanding that the RPI 3 GPIO is supposed to be backwards compatible with RPI 2. If this is the case than it should just be a matter of defining the RPI 3 and using the RPI 2 PinMappings. I don't own a RPI 3 so I am not able to test this otherwise I would do the work. If someone has a RPI 3 and wants to give it a shot look at the PinMapping.cs located in Raspberry.IO.GeneralPurpose

OddmarDam, 2016-05-22:
I have a RPI 3. I suppose it's the BoardRevision i am looking for ?!

jyksnw, 2016-05-22:
I just had a deeper look at this. Looks like PinMapping has already been updated to account for RPI 2 and RPI 3, however GpioConnectionSettings.cs had not. I added a HACK into GpioConnectionSettings.cs that might potentially fix the RPI 3 I2C issue. This isn't a long term fix, so if someone can test it and get back to me I will gladly work on a more permanent solution.

You can find the fix/hack in my [fork of raspberry-sharp-io](https://github.com/jyksnw/raspberry-sharp-io/tree/rpi3_i2c). Be sure to checkout the rpi3_i2c branch after cloning the fork.

OddmarDam, 2016-05-23:
Here is what i get when i test the I2C on a RPI3::

Unhandled Exception:
System.NotSupportedException: Connector pintout Unknown is not supported
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase (ProcessorPin
sdaPin, ProcessorPin sclPin) <0x74a5e610 + 0x001f4> in <filename unknown>:0
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver..ctor (ProcessorPin
sdaPin, ProcessorPin sclPin) <0x74a5e268 + 0x0005b> in <filename unknown>:0
  at Test.Components.Ssd1306.MainClass.Main (System.String[] args)
<0x76abdf50 + 0x00047> in <filename unknown>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.NotSupportedException: Connector
pintout Unknown is not supported
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase (ProcessorPin
sdaPin, ProcessorPin sclPin) <0x74a5e610 + 0x001f4> in <filename unknown>:0
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver..ctor (ProcessorPin
sdaPin, ProcessorPin sclPin) <0x74a5e268 + 0x0005b> in <filename unknown>:0
  at Test.Components.Ssd1306.MainClass.Main (System.String[] args)
<0x76abdf50 + 0x00047> in <filename unknown>:0

msanchez71, 2016-05-23:
jyksnw,

I would be happy to send you a RPi 3 if you would like.  just let me know and we can work out the logistics.  I will also do some more testing this week as I get a chance.

Thanks...

jyksnw, 2016-05-23:
@OddmarDam thanks for posting back. Assuming that the error was caused while using my rpi3_i2c branch; The error "System.NotSupportedException: Connector
pintout Unknown is not supported" might indicate that the ConfigurationManager isn't able find the gpioConfiguration, at least that is my assumption. I would need a RPi 3 to trace down the root cause of the error and potentially fix it.

@msanchez71 Thank you so much for the offer. I have been wanting a reason to pull the trigger to get myself a RPi 3 so maybe now is a good time. I wouldn't want to put you out of your RPi 3 while I tried to troubleshoot and this I2C issue.

jyksnw, 2016-05-24:
I have been trying to re-situatiate myself with this project this evening as it has been about a years since I contributed to it. I have found that It has also been about a year since there has been any major contribution to this library. With the RPi 2 and RPi 3 having support for Windows 10 IoT I almost hate to suggest that one looks there for a more stable and viable C# based solution. Looks like they have the drivers for the I2C stuff in Win 10 IoT. Have a look at [Windows 10 IoT](https://developer.microsoft.com/en-us/windows/iot) and their [GitHub](https://github.com/ms-iot). I did pull in the two pending Pull Requests into my fork and I will try to keep working on this project as my time permits, but I can't make any promises.

psymonwhite, 2016-05-24:
I've been using the Windows 10 IoT for a while now, and although it works well enough, it's worth pointing out that it is not a full .Net implementation. 

Many of the basic system libraries are missing and you have to look up the Windows Universal Platform way to do things. 

I went over to this because the implementation is much smoother and the remote debugging is very easy to setup.

OddmarDam, 2016-05-24:
I also tried the Windows iot but found a lot of the "old" ways are different. TCPClient is not there, last time i checked.
So please keep up the good work :)

jyksnw, 2016-05-24:
Thanks for the feedback on Windows 10 IoT. I will gladly keep working on this, I don't have write privileges to this repository so I am going to focus on implementing any fixes on my fork of the project.

StarlightNL, 2016-05-24:
After a long time I decided to respond to this thread. Yes Raspberry-sharp-io isn't updated for a long time. You can get I2C working on a Pi3 by hacking a bit in some classes but because of some unknown reason it is really buggy and unstable and I just remade my program in Python.
I couldn't(and wouldn't) use Windows 10 IoT because the network support is not that good(Can't even run ASP) and the whole coding of it is not what I'm used to.  
I hope the devs of Raspberry-Sharp-IO are gonna update the code because it was really usefull. But till that time the choice for you guys could be either ditch Raspbian and use Windows 10 IoT or ditch .NET and try to use Python.

R.I.P. Pi 3. You served me well till a power addon board caused a malfunction.

ebezine, 2016-05-25:
Hi,
Sorry for the late reply, the project is a bit frozen currently due to high pressure at work. Unfortunately I still haven't bought a Raspberry Pi 3 hence I'm not able to test.

As far as I understand the issue, RPi 3 pinout is not correctly detected in GpioConnectionSettings. I just had a look and it seem that RPi 2 is not handled (although it might be compatible with RPI B+ ?)

I'll try to buy a RPi 3 soon in order to make some tests and fix this issue.

Thanks for your support... and patience !

ebezine, 2016-05-25:
Actually Pinout detection is made in Raspberry.Board (i.e in Raspberry.System assembly)

ebezine, 2016-05-25:
I just merge https://github.com/raspberry-sharp/raspberry-sharp-system/pull/3
Does this help to solve the issue ?

erwinkramer, 2016-05-31:
@ebezine & @jyksnw ,

I tested both of your solutions on the Pi 3 and both of them resulted in the original error message "Connector pintout Unknown is not supported"

jyksnw, 2016-05-31:
@erwinkramer 

Thanks for the feedback. This might be a silly questions, but did you try both of our changes at the same time? If so, please try @ebezine change independent from the changes I made, my changes are only a quick and untested hack as I don't have a RPI 3 to test with.

erwinkramer, 2016-05-31:
@jyksnw 

I didn't try both changes at the same time, i'm not sure what parts from the codebases i should get in order to get both changes in 1 package.

jyksnw, 2016-05-31:
@erwinkramer,

Like I said it was probably a silly question. Please don't try using both @ebezine change and my change at the same time. I am going to delete my branch as it was just a quick hack and it didn't fix the issue.

Thanks again for the feedback

ebezine, 2016-06-01:
Hello,

I've received today brand new RPi 3 and RPi Zero. I've updated raspberry-sharp-system repository to add support for Pi Zero (Pi 3 was already supported).

I've also pushed new versions of both nugets to ensure new boards support is propagated to raspberry-sharp-io too.

This should solve issue

xlevel, 2016-08-09:
@ebezine I have built the latest source and tried the SSD1306 example. I get the same exceptions as @OddmarDam above:
```
System.NotSupportedException: Connector pintout Unknown is not supported
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase (ProcessorPin sdaPin, ProcessorPin sclPin) <0x76892b50 + 0x0021c> in <filename unknown>:0 
  at Raspberry.IO.InterIntegratedCircuit.I2cDriver..ctor (ProcessorPin sdaPin, ProcessorPin sclPin) <0x76892818 + 0x0005b> in <filename unknown>:0 
```

Freaky-Styley, 2016-08-21:
Hi! 
I'm trying to do I²C on a Pi 3 as well, but it still doesnt work.  At the current release of sharp-io i found @ Raspberry.IO.InterIntegratedCircuit.I2cDriver.GetBscBase() that there are supported the ConnectorPinouts of "Rev1", "Rev2" and "Plus". Is "Rev3" missing or am i using the wrong release?
In this method, the "GpioConnectionSettings.ConnectorPinout" returns unknown. Further i can't find the "BCM 2837" specified, just BCM2836/BCM2835 for example.

Do you think its also possible to add support for BananaPi?

Thanks for the great work!

xlevel, 2016-08-29:
I have got the SSD1306 test application working on a Raspberry Pi 3. It turned out to be an out the the version of Raspberry.System.dll referenced is currently 2.0.0 rather than 2.1.0. Updating this reference makes it work.

xlevel, 2016-09-01:
This issue should be now fixed after [this commit](https://github.com/raspberry-sharp/raspberry-sharp-io/commit/638f67fc8466d53d60168f2e896e51d77ed6823b) was merged in.

closed 2016-09-01 by raspberry-sharp

erwinkramer, 2016-09-07:
Thanks @xlevel 

I just confirmed it working as well in combination with the FEZ HAT.

bharatmittal1992, 2018-03-12:
@xlevel could you please provide me sample project which have how i can read/write data in gpio in C#