raspberry-sharp/raspberry-sharp-io #62: Why is InterIntegratedCircuit missing on NuGet?

MihaMarkic, 2016-06-18:
Why is InterIntegratedCircuit missing on NuGet?

xlevel, 2016-08-11:
Same here. Is this something that someone could take on for the project?

JTrotta, 2016-08-21:
You were right. Next week I will fork the project with some corrections under debugging, and port it on Nuget. Please wait...

paultechguy, 2017-01-13:
bump.  Any chance of the updated nuget with InterIntegratedCircuit and Raspberry.IO.Components.Controllers.Pca9685?

JerseyTechGuy, 2017-04-07:
It's now April 2017 and InterIntegratedCircuit is still missing.  Might be time to remove I2C from the description. Very misleading.

guehmann, 2017-05-10:
Today wanted to include the SPI Assembly in my project via nuget, but Raspberry.IO.SerialPeripheralInterface is also missing. Using Mono on Ubuntu but it's the same on Windows with Visual Studio. Do I'm something wrong? So i cloned the project and build it with Mono, build succeeded.

paultechguy, 2017-05-10:
Yes, had same issue so once I built it, I built my own nuget (local) and
use that with all the needed assemblies.

guehmann, 2017-05-10:
@paultechguy perhaps you have an example of a working implementation of MIfare RC522 with a SPI?
Thanks in advance.
Jens

paultechguy, 2017-05-10:
@guehmann, sorry, do not have that done.  I ended up packing the sharpio libs with my app...all the ones I needed.

**GPIO Examples in C#**
https://github.com/paultechguy/RaspberryPi.DotNet.GpioExamples

**Extensible GPIO Web Services in C#**
https://github.com/paultechguy/RaspberryPi.DotNet.GpioWeb

guehmann, 2017-05-10:
@paultechguy thank you very much, that's great. think will help me.
Regards Jens.