raspberry-sharp/raspberry-sharp-io #72: How to convert this lib to .net Core and use it into Windows IOT on RPI3

pdsavard, 2016-10-28:
Someone use this lib in windows IOT program? 
How I can compile it with core.net. My projecr is build for Windows IOT univesall app Windows 10 build 10586.

Thanks

ScramOnline, 2017-02-05:
https://github.com/ScramOnline/raspberry-sharp-io i create this fork, works over ubuntu

donker, 2017-03-01:
I'm also interested in a .net core version. I started on this but got bogged down in conversion of code I'm not familiar with. So gave up after a while until either someone else does this or I familiarize myself with this code and the conversion process to dotnet core.

donker, 2017-03-01:
Actually the aforementioned ScramOnline fork seems to have nailed it. Sorry to waste anyone's time.

MaitreDede, 2017-05-02:
Hi @ScramOnline 

Thanks for your port.
Do you plan to publish your fork to nuget ?
Do you think you can manage to upgrade the lib to be usable also on Windows 10 IoT ?

ScramOnline, 2017-05-08:
On Wed, May 3, 2017 at 12:43 AM, Damien DALY <notifications@github.com>
wrote:

> Do you plan to publish your fork to nuget ?
> Do you think you can manage to upgrade the lib to be usable also on
> Windows 10 IoT ?
>

​Do you plan to publish your fork to nuget ? I would like to do it in the
next month.
Do you think you can manage to upgrade the lib to be usable also on Windows
10 IoT ?​ No i don't know

-------------------------------------------------
Atentament Ramon Balaguer
<rba.cat@gmail.com>rba.cat@gmail.com
676719793

nzain, 2017-07-13:
Unfortunately, this is outdated. The fork uses deprecated xproj structure and visual studio solution (I'm on linux, using VS Code). I'll see if I'm able to make it work on dotnet core 2.0 and vs code.