raspberry-sharp/raspberry-sharp-io #82: ANyone have a work copy

iHover01, 2017-01-30:
Ive trying to get this to compile on mono for two days now and between the third party libs and being new to mono it's just not happening. Does anyone have the dll's for this and any other files required. Id be will to PP someone a twenty just to get back to working on my actual project rather than spending more time beating my head against this particular wall

nahueltaibo, 2017-02-01:
I needed to download the nuget extension, and then, one by one remove all the nuget packages references and add them again... Also removed the test projects from the solution, because they had issues too, and I didnt need them.

Good luck!

moelski, 2017-02-16:
@iHover01: 
If you still have problems just send me an email. I compiled rasp-sharp-io two days ago using Visual Studio. Can check it with Monodevelop, too. It worked like a charm for me. 
-> dominik <ätt> logview <dot> info

Dominik