raspberry-sharp/raspberry-sharp-io #86: Always getting an exception GpioConnection: Object reference not set to an instance of an object

Asshen, 2017-05-06:
Hey.

I have a Raspberry Pi 3 B. I did some GPIO development using Python, but I would love to use C#.
I set up a small project (Raspberry.IO references are OK):

```
using System;
using Raspberry.IO.GeneralPurpose;

namespace Lichtje1
{
    class Program
    {
        public static void Main( string[] args )
        {
            var led1 = ConnectorPin.P1Pin15.Output();

            var connection = new GpioConnection( led1 );

            for( var i = 0; i < 100; i++ )
            {
                connection.Toggle( led1 );
                System.Threading.Thread.Sleep( 250 );
            }

            connection.Close();
        }
    }
}
```

I keep getting the following error no matter if I compile on Visual Studio (on Windows) or MonoDevelop (on Raspberry Pi):

```
Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0
  at Lichtje1.Program.Main (System.String[] args) [0x00011] in <fb46ca8bbdcc45f0952b44480e70cfba>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0
  at Lichtje1.Program.Main (System.String[] args) [0x00011] in <fb46ca8bbdcc45f0952b44480e70cfba>:0
```

Thank you for any help.
Kind regards.
Kris.

Arakis, 2017-05-07:
i have a similar problem:

```
System.NullReferenceException: Object reference not set to an instance of an object
at Raspberry.IO.GeneralPurpose.GpioOutputBinaryPin..ctor
```

Its since the upgrade from kernel ARCH Linux 4.4 to 4.9

Bulinlinbu, 2017-05-30:
So do I

It seems that "Driver" is null GeneralPurpose > GpioConnection.cs line 488 :
`Driver.Allocate(configuration.Pin, configuration.Direction);`

Bulinlinbu, 2017-05-30:
OK, il think the problem is from Raspberry.System API, it doesn't reconize the CPU of this version of PI.

If I do this :
`Raspberry.Board.Current.IsRaspberryPi; // False`

See this request :
[(https://github.com/raspberry-sharp/raspberry-sharp-system/issues/6)]

Bulinlinbu, 2017-05-30:
Right, it works.
First : in Raspberry.System, add the BCM2835 CPU 
see this post : [htt://raspberry-sharp/raspberry-sharp-system#6](raspberry-sharp/raspberry-sharp-system#6l)

Second : In Raspberry.IO.GeneralPurpose, edit the function :
`private static uint GetProcessorBaseAddress(Processor processor)
        {
            switch (processor)
            {
                case Processor.Bcm2708:
                    return Interop.BCM2835_GPIO_BASE;

                case Processor.Bcm2709:
                **case Processor.Bcm2835:**
                    return Interop.BCM2836_GPIO_BASE;

                default:
                    throw new ArgumentOutOfRangeException("processor");
            }
        }`

NeededAnUnusedName, 2017-07-02:
Hi, can anyone please repeat the steps (as simple as possible ^^) to solve this problem? There is a dead link. I found the function above but there are so many build errors... Is there a simpler solution, or can I maybe update to a fixed version with nuget?

Bulinlinbu, 2017-07-02:
Link not dead anymore

NeededAnUnusedName, 2017-07-02:
Thanks, maybe fix the other link too, I still cant reach it.
But it helped a lot, thank you very much! =)

JohnRusk, 2017-10-22:
A note for others who find this page: it's recommended to use the revision number now, for identifying the processor.  See https://github.com/raspberrypi/linux/issues/2008 and https://github.com/raspberrypi/firmware/issues/705 

Also, I wonder if even older Pi's report Bcm2835 as the Hardware value in `/proc/cpuinfonow`. (The change, after all, was in the OS kernel, not the actual hardware). If that's the case, the both old and new Pi's may be reporting the same value, in which case the fix suggested by @Bulinlinbu above will not be safe on older Pi's.  I have no way of testing this tho, since I only have a new one.

However, if you _know_ you're only running on a RPi 3, then @Bulinlinbu's change is safe enough. 

Finally, test on my RPi3 indicates that this quick and dirty workaround also works, if you _know_ you are running on a RPi 3:

`
            Dictionary<string, string> boardSettings = (Dictionary<string, string>)
                                                        Board.Current.GetType()
                                                       .GetField("settings", BindingFlags.Instance | BindingFlags.NonPublic)
                                                       .GetValue(Board.Current);
            boardSettings["Hardware"] = "BCM2709";
`

I wouldn't advocate using that for any long-term purpose, but as a quick workaround until R#.IO, for users who _know_ they are on RPi3, it seems to work.

JTrotta, 2018-08-07:
Raspberry.IO.GeneralPurpose3 has been disocontinued.
Try [https://github.com/JTrotta/RaspberrySharp](url)

piibet, 2018-12-16:
Please make sure you have **enabled** all **Interfaces** in **Rasperry Pi Configuration**.
I forgot to do that, and it gave me same error.