raspberry-sharp/raspberry-sharp-io #87: Using SPI with Mifare RC522

guehmann, 2017-05-10:
Does anybody has an example to read RFID tags from an Mifare RC522 with Raspberry.IO.SerialPeripheralInterface? I have a working example in python from https://github.com/ondryaso/pi-rc522 but i need to implement the complete logic into my mono application.

Thanks in advance.
Regards Jens.

awaescher, 2017-07-29:
Hey @guehmann, did you make an progress? Same thing over here ...

guehmann, 2017-07-29:
Hi, no sorry in this situation i have no progress. I totally switched to Windows 10Iot, so I can use the Visual Studio Environment. My Solution is now a Windows UWP. Within this it's no problem connecting to RFID-Mifare, there a a lot of samples on git.

awaescher, 2017-07-29:
Okay, thank you for that hint.