raspberry-sharp/raspberry-sharp-io #91: Repeated Start Condition - How to code?

moelski, 2017-07-21:
Hi @all, 

today I tried to implement the AS3935 Lightning Chip in my application. But I run into problems reading the Registers separately.  

My first attempt was this: 
`i2cConnection.Write(0x01);
 byte res = i2cConnection.ReadByte();`
This should read the Byte from Register 0x01. But I alsways get the data from Register 0x00. 

My second try was this Code: 
`                var readAction = new I2cReadAction(1);
                
                i2cConnection.Execute(new I2cTransaction(new I2cWriteAction(0x01), readAction));
                Console.WriteLine($"ergebnis : {readAction.Buffer[0]}");`
But this also did not work. 

If I read a bulk of data then it works. But that is not a good solution because you have to read to many data at a time. 

For me it look like a Problem with the repeated start condition (which is needed for the AS3935 device). I found some notes about changes for the repeated start. But how can I use it? 

Any hint would be much appreciated !

regards Dominik

JTrotta, 2017-09-25:
Got the same problem with this code:
`            I2cWriteAction writeAction = new I2cWriteAction(register);
            I2cReadAction readAction = new I2cReadAction(new byte[byteCount]);
            I2cTransaction transaction = new I2cTransaction(writeAction, readAction);
            Execute(transaction);

            return readAction.Buffer;`

JTrotta, 2017-10-21:
Have a look to my new number libraries. Now repeated start is workinf

moelski, 2017-11-02:
Hi !
Which libraries did you mean? 
I can´t find any new stuff. 

regards Dominik

JTrotta, 2017-11-02:
This one:
https://github.com/JTrotta/raspberry-sharp-io/tree/master/Raspberry.IO.InterIntegratedCircuit

I rewrote it, and it works now. Try nuget packages.
https://www.nuget.org/packages/Raspberry.IO.InterIntegratedCircuit3