raspberry-sharp/raspberry-sharp-io #96: Object Refference Null on new GpioConnection(InputPinConfiguration[])

FalcoGer, 2018-02-05:
I'm using a Raspberry Pi 2 B+ with Raspbian Jessie and am writing a Windows Forms application that's running on Mono.
The app runs fine. It is supposed to run on an [LCD Display](https://github.com/watterott/RPi-Display).
The display works fine too. According to the specs it has 3 switches connected to GPIO Pins 21, 22 and 27.
I'm a Programmer, not a hardware engineer. I think I figured out that those GPIO Pins are on the real (physical) pins 11, 13 and 15. (Why they are numbered differently is beyond me.)
In the application I want to fire a function when the buttons are pressed.

Here is what I got:
in Form class:
```
public partial class FormMyApp : Form
{
...
        private InputPinConfiguration SW1_GPIO17 = null;        // GPIO 17 (Pin 11)
        private InputPinConfiguration SW2_GPIO27 = null;        // GIPO 27 (Pin 13)
        private InputPinConfiguration SW3_GPIO22 = null;        // GIPO 22
        private GpioConnection gpioConnection = null;
...
```
in Form ctor:
```
public FormMyApp()
{
...
    try
            {
                this.SW1_GPIO17 = ConnectorPin.P1Pin11.Input().Switch().Enable();     // GIPO 17
                SW1_GPIO17.Name = "Switch 1";
                SW1_GPIO17.OnStatusChanged<InputPinConfiguration>(x => SwitchPressed(x, 1));
                this.SW2_GPIO27 = ConnectorPin.P1Pin13.Input().Switch().Enable();     // GPIO 27
                SW2_GPIO27.Name = "Switch 2";
                SW2_GPIO27.OnStatusChanged<InputPinConfiguration>(x => SwitchPressed(x, 2));
                this.SW3_GPIO22 = ConnectorPin.P1Pin15.Input().Switch().Enable();     // GPIO 22
                SW3_GPIO22.Name = "Switch 3";
                SW3_GPIO22.OnStatusChanged<InputPinConfiguration>(x => SwitchPressed(x, 3));
                PinConfiguration[] pinConfig = new PinConfiguration[] { SW1_GPIO17, SW2_GPIO27, SW3_GPIO22 };
                Console.WriteLine("GPIO Config:");
                Console.WriteLine(pinConfig);
                Console.WriteLine("PinConfigs:");
                foreach (PinConfiguration pc in pinConfig)
                {
                    Console.WriteLine(pc);
                    Console.WriteLine("Name: " + pc.Name );
                    Console.WriteLine("Pin: " + pc.Pin);
                    Console.WriteLine("Action: " + pc.StatusChangedAction);
                    Console.WriteLine("Direction: " + Enum.GetName(typeof(PinDirection), pc.Direction));
                    Console.WriteLine();
                }
                // above code prints okay
                // this.gpioConnection = new GpioConnection(SW1_GPIO17); // <-- throws exception
                // this.gpioConnection = new GpioConnection(pinConfig); // <-- throws exception
            }
            catch (Exception e)
            {
                logWriter.WriteLine(DateTime.Now.ToString(TIMEFORMAT) + ": EXCEPTION. SEE ERROR LOGS FOR DETAILS.");
                ErrorLog.WriteException(e);
                throw new Exception("Could not create gpio connection", e);
            }
...
```
and
```
        private void SwitchPressed(bool x, int switchId)
        {
            Label_TestLabel.Text = "Switch " + switchId + " is now " + (x ? "High" : "Low");
        }
```

I don't know what I'm doing wrong. And there isn't any documentation on how to read from an input and fire on events.
Unfortunately I'm developping on windows and can't debug the code in VS because the windows machine doesn't have the GPIO hardware, of course.

Here is the exception I got:
EXCEPTION

Exception time: 05.02.2018 10:38:03.3573
Exception message: Object reference not set to an instance of an object
Exception source: Raspberry.IO.GeneralPurpose
Help link: 

================================================================================

Stack Trace:
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0 
  at myApp.FormMyApp..ctor () [0x001ce] in <e622084571bb433199911dfa030a90f2>:0 
================================================================================

FalcoGer, 2018-02-05:
I have found another solution in
https://github.com/cypherkey/RaspberryPi.Net.git
Using http://www.airspayce.com/mikem/bcm2835
I can use c# to set the resistors to pull up and make my switches work.
I use a 100ms timer on the win form to update the switch status.

Dip11, 2018-08-07:
this things happens because of the processor, if you are using pi3 then the processor is bcm2835 which is not implemented in Raspberry.IO.GeneralPurpose package. to solve this issue try to install Raspberry.IO.GeneralPurpose3 -Version 3.1.1.
Link: https://nuget.org/packages/Raspberry.IO.GeneralPurpose3/