raspberry-sharp/raspberry-sharp-io #104: Using raspberry-sharp-io with raspberry pi3 model b+ and last version of mono..

hcastanedaguat, 2018-07-30:
Hi..
  using the latest download of this library and used with the raspberry pi3 model b+ and latest version of mono we are getting errors like the exposed below 
Can any help me to fix this issue..

Thanks

pi@raspberrypi:~/Desktop/SharedInWin/testmono $ gksudo mono DesktopApp1.exe
System.NullReferenceException: Object reference not set to an instance of an object
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0 
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0 
  at DesktopApp1.Form1.Form1_Load (System.Object sender, System.EventArgs e) [0x0002c] in <c0e2b4966ced45e98cf196a13a0d692e>:0 
  at System.Windows.Forms.Form.OnLoad (System.EventArgs e) [0x00022] in <d5b72d15d4f7424c8a1538e3f19ec2e3>:0 
  at System.Windows.Forms.Form.OnLoadInternal (System.EventArgs e) [0x00029] in <d5b72d15d4f7424c8a1538e3f19ec2e3>:0

grimreaper38uk, 2018-08-15:
I'm guessing folks are going to have to see some of your source code before you get a response to this...

negusm, 2022-06-10:
Same issue:
Command:
sudo mono PiFlambeDaemon.exe
Error: 
`Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object.
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0
  at PiFlambeDaemon.Program.Main () [0x00009] in <e309ff0ee38543af83093bfd1121b593>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object.
  at Raspberry.IO.GeneralPurpose.GpioConnection.Allocate (Raspberry.IO.GeneralPurpose.PinConfiguration configuration) [0x00073] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection.Open () [0x00036] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x000f8] in <39616ab275014559a5b975fe859e9ad9>:0
  at Raspberry.IO.GeneralPurpose.GpioConnection..ctor (Raspberry.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <39616ab275014559a5b975fe859e9ad9>:0
  at PiFlambeDaemon.Program.Main () [0x00009] in <e309ff0ee38543af83093bfd1121b593>:0`
Code:
`using System;
using Raspberry.IO.GeneralPurpose;

namespace PiFlambeDaemon
{
    internal class Program
    {
        static void Main()
        {
			InputPinConfiguration btn1 = ConnectorPin.P1Pin18.Input();
			GpioConnection connection = new GpioConnection(btn1);
		}
    }
}
`

negusm, 2022-06-10:
Nevermind...I see that there is a newer version here:
https://nuget.org/packages/Raspberry.IO.GeneralPurpose3/

This solved my problem.