raspberry-sharp/raspberry-sharp-io #112: Does this lib work on RPI4b?

CsharpAndVB, 2020-08-28:
Hi there,
I'm new to mono & the Pi so bare with...

Does this library work on the Pi 4b?
If not, are there any that do?

Thank you

CsharpAndVB, 2020-09-25:
Can anyone help!?

I can't get this working on a 4b. Should I use a Rpi 3 instead?

waiter21, 2021-01-11:
I think there is an issue with it identifying the correct processor.  I'm having the same problem using a Rpi 3B+. I be;lieve this is the issue:

https://github.com/raspberry-sharp/raspberry-sharp-io/issues/88

I've been trying to make the changes that user https://github.com/michaltalaga suggested,  but I'm not having any luck.

n8allan, 2021-02-15:
This project appears to have been abandoned.  Here's what looks to be an active fork:
https://github.com/genemars/raspberry-sharp-io